microsoft/CogACT #1: Inquiry Regarding Batch Size and GPU Memory Usage on A800 vs. A100 GPUs

CyanChew, 2024-12-09:
Dear CogAct Team,

Thank you so much for open-sourcing your code and providing detailed documentation. 

I’ve been trying to replicate your experimental setup as described in your paper and code. According to your reported settings, you were able to train with a batch size of 32 using an 80GB A100 GPU. However, when I attempt to run the same code and model on my 80GB A800 GPU, I encounter out-of-memory issues even when I significantly reduce the batch size to 2.

Eureka6174, 2024-12-10:
We use Fully Sharded Data Parallel (FSDP) to distribute the model's parameters, gradients, and optimizer states across multiple GPUs. Therefore, the code must be run on a server equipped with multiple GPUs. If the server has 8 GPUs with 80GB of memory each, you can set the batch size to 32. With fewer GPUs on the server, the batch size should be reduced accordingly. Note that it may not be possible to run the code with a single GPU.

CyanChew, 2024-12-10:
thank you for your reply, now I close the comment

closed 2024-12-10 by CyanChew