microsoft/CogACT #4: Realman robot dataset collection

zhujingsong, 2024-12-20:
Hi, I want to reproduce your experiment on Realman robot since our lab has a similar one.
1. Are there any plans to publish your data collected on Realman?
2. Since I am new to teleoperation, could you share the code for teleoperation and data recording.
Thanks!

arnoldland, 2024-12-24:
Thank you for your interest! We do not plan to release the code and data related to teleoperation. This is because it heavily depends on the hardware used, such as different hardware devices, assembly methods, APIs, and control methods. Our teleoperation code is unlikely to be directly usable in other labs; you need to set up your own teleoperation system and record data. Additionally, the extremely small amount of data we collected in narrow environments is also unlikely to be directly usable in the setups of other labs. 

However, we can provide some useful references for setting up teleoperation, such as the [OpenTeleVision system](https://github.com/OpenTeleVision/TeleVision); documentation and tutorials for using Meta Quest Building Blocks (if you are also using): https://developers.meta.com/horizon/documentation/unity/bb-overview, (unofficial)https://www.youtube.com/watch?v=I8_BYPruDMk.

closed 2024-12-31 by arnoldland

zhujingsong, 2025-01-03:
> Thank you for your interest! We do not plan to release the code and data related to teleoperation. This is because it heavily depends on the hardware used, such as different hardware devices, assembly methods, APIs, and control methods. Our teleoperation code is unlikely to be directly usable in other labs; you need to set up your own teleoperation system and record data. Additionally, the extremely small amount of data we collected in narrow environments is also unlikely to be directly usable in the setups of other labs.
> 
> However, we can provide some useful references for setting up the teleoperation, such as the [OpenTeleVision system](https://github.com/OpenTeleVision/TeleVision), documents and tutorials for using Meta Quest Building Blocks (if you are also using): https://developers.meta.com/horizon/documentation/unity/bb-overview, (unofficial) https://www.youtube.com/watch?v=I8_BYPruDMk.

Thanks for your reply!
I have noticed that the raw output action of CogACT is the relative transition and rotation delta_x, delta_y, delta_z, delta_r, delta_p, delta_y. Does it mean the policy is kind of cross-embodiment？？？Like UMI Gripper？？？What really matters is the IK caculation？？？

arnoldland, 2025-01-06:
Yes, you're right. We predict the 6D pose of the end effector instead of the joint angles, which makes it more general for different embodiments. Obtaining the joint angles requires IK calculations. Additionally, our strategy is indeed a cross-embodiment policy.