microsoft/CogACT #5: how many epochs it take to train cogact from scratch

LiHaoHN, 2024-12-22:
Hello, thank you for your work and open-source code! I want to ask how many epochs it takes to train cogact from scratch, It is written in code that oxe_magic_soup_plus_minus configures 100 epochs, but I think it is very time-consuming (more time-consuming than training openvla?), you mentioned in your paper that you trained cogact 135K iteration, and your data is 22.5 million frames,with batchsize 256, does this mean you trained 135k/(22.5 M/256)= 1.536 epoch

arnoldland, 2024-12-24:
Thank you for your attention!
1. In our config, we set epochs=100, but we manually terminated the training well before reaching this length of training time. The model we actually released was trained for 135K steps as stated in our paper. Due to the sampling mechanism of the dataloader, different datasets do not completely go through one full pass at the same time. However, The number of steps we used resulted in the main datasets (such as fractal20220817_data) going through approximately 2 epochs.

2. Good question! In fact, determining whether the robot policy has reached the optimal training step is a relatively difficult problem. Since we use diffusion loss to model the action sequences, there isn't a metric like token accuracy in OpenVLA to help us determine if the training should finish. For the pretraining stage, we suggest you can terminate the training when the average (Time Weighted EMA) training loss is about 0.05. As for the finetuning stage, we observed that training for approximately 15 epochs already yielded satisfactory performance. We have not yet thoroughly tested models finetuned for a longer duration in the real-world robot.

LiHaoHN, 2024-12-24:
Get it! Thank you for your reply!

closed 2024-12-24 by LiHaoHN

LiHaoHN, 2024-12-24:
> Thank you for your attention!
>
> 1. In our config, we set epochs=100, but we manually terminated the training well before reaching this length of training time. The model we actually released was [...]

I have another question. Under oxe_magic_soup_plus_minus ratio, what is the total length of the dataset? Is it consistent with the number of frames, like the 22.5M frames written in the paper, or is it smaller than this number?

arnoldland, 2025-01-06:
We have solved this issue through offline communication. :)