microsoft/CogACT #6: No module named 'prismatic'

Winston-Yuan, 2024-12-25:
CogACT/vla/load.py

IDs, mappings to paper experiments, and short descriptions), as well as for loading models (from disk or HF Hub).
"""

import json
import os
from pathlib import Path
from typing import List, Optional, Union

from huggingface_hub import HfFileSystem, hf_hub_download

from prismatic.conf import ModelConfig
from prismatic.models.materialize import get_llm_backbone_and_tokenizer, get_vision_backbone_and_transform
from prismatic.models.registry import GLOBAL_REGISTRY, MODEL_REGISTRY
from prismatic.models.vlms import PrismaticVLM
from prismatic.overwatch import initialize_overwatch

from vla import CogACT

######################

the 'prismatic' module is not included in the repo

arnoldland, 2024-12-27:
Please install the dependencies by ``pip install -e .``, and the 'prismatic' module will be included. Related to #2.

arnoldland, 2024-12-27:
We have modified the 'prismatic' module in [this repository](https://github.com/arnoldland/openvla), and it is included as a dependency in [pyproject.toml](https://github.com/microsoft/CogACT/blob/main/pyproject.toml). Please feel free to make further modifications to the 'prismatic' module in the provided repository if necessary.