microsoft/CogACT #7: Evaluation on SimplerEnv

Koorye, 2024-12-31:
Thanks for your work! 
I would like to know if the evaluation on simplerenv needs to do fine-tune, or just use the pre-trained weights on OXE?

arnoldland, 2024-12-31:
Thanks for your attention!
Our evaluation on SimplerEnv only requires the pre-trained weights on OXE, without the need for any additional fine-tuning. Moreover, the pre-trained weights we provide have been trained solely on OXE and have not undergone any specific fine-tuning. We use the same data and training phases as other models to ensure a completely fair comparison. However, if you continue to fine-tune the pre-trained models on a specific dataset, you will achieve even better results.

Koorye, 2024-12-31:
Thank you! It helps me a lot.

closed 2024-12-31 by Koorye