microsoft/CogACT #12: Where is the cognition token input to the language decoder？

feifei-Liu, 2025-01-17:
Hi，
Thank you for your excellent work.

[image]

In my learning CogACT, I noticed that there was a cognition token on the diagram that was sent to language decoder. But I didn't find the definition of the cognition token in the network implementation, only found its output https://github.com/Microsoft/cogact/blob/main/vla/cogactvla.py # l145. Can you help me find out how the cognition token defines，before it input to the language decoder? 

Thank you very much！

feifei-Liu, 2025-01-21:
Hello，

After read the https://github.com/arnoldland/openvla/blob/5603207085d55148682e2a35b868ad77d7b42ece/prismatic/models/backbones/llm/prompting/base_prompter.py#L40 and https://github.com/microsoft/CogACT/blob/main/vla/cogactvla.py#L431, I guess if the cognition token is the 'EOS token' ?

Besides,  if the cognition token is the 'EOS token',  why should we insert the empty token ('') https://github.com/microsoft/CogACT/blob/main/vla/cogactvla.py#L435?

arnoldland, 2025-01-21:
Apologies for the delayed response. We use a special token to represent the cognition token, and in this case, we actually use the 'EOS' token. Of course, you can use any other special token that doesn't affect the training. 

The inclusion of an empty token ('') aims to align the inputs during training. During the training phase, when creating the chat, an empty token ('') is added at the end of the conversation. You can also remove this empty token during training. As long as the inference and training phases are consistent, its presence or absence is not important. You can check the 'input_ids' to obtain a clearer understanding.

feifei-Liu, 2025-01-22:
Thank you very much! It helps me a lot!

closed 2025-01-22 by feifei-Liu