microsoft/CogACT #14: Question About Action Modes and Data Collection for Reproducing Your Work

capret, 2025-01-24:
First of all, I want to say that your work is incredible and very interesting! I’m excited to reproduce it using a custom collected dataset. Although I’m new to robotics, I’m very enthusiastic about learning and exploring this field !

When I reviewed the simulation you provided (ManiSkill2_real2sim), I noticed that the action mode in the code includes four types: `base`, `ee`, `ee_align`, and `ee_align2`. I also observed that some robots, such as WidowX, use `ee_align2`, while others like the Google robot use `ee_align`.  

If I plan to collect data for fine-tuning on your work, could you clarify which action mode I should calculate from the end-effector (EE) pose? Or does it not matter because your work appears to mix datasets, and all action modes are acceptable for training?  

Thank you so much for your time and guidance!

arnoldland, 2025-02-06:
Thank you for your interest! ``ee_align`` and ``ee_align2`` are just methods used by ManiSkill to control different robots. The model only needs to predict the delta of the end-effector's 6D pose, ensuring that the representation is aligned during the finetuning and testing time. Subsequently, the model's predictions are converted into the action format required to control the specific robot.

capret, 2025-02-06:
Thank you very much for your response; it's great help. Actually, I'm confused about the definition of "delta 6D pose." If I conduct experiments in a simulation environment and create my own simulation dataset, when I record a list of the end-effector's 6D poses for a task, would the action be directly calculated as t1 - t0?

arnoldland, 2025-02-06:
If you record the absolute values of the 6D pose, you need to calculate the delta of the 6D pose. Here, the 6D pose consists of XYZ (3) + Roll-Pitch-Yaw (3). We define delta translation as the difference in translation: $\Delta t = t_n - t_{n-1} $. The definition of delta rotation involves first calculating the rotation matrices $R_n$ and $R_{n-1}$, then computing $\Delta R = R_n R_{n-1}^{-1}$. Finally, convert the rotation matrix $\Delta R$ back to Euler angles.

capret, 2025-02-06:
Thank you very much! I’m very appreciate it. It seems that the delta pose you described should work with the “ee_align” method we mentioned before. Now I can understand.

closed 2025-02-06 by capret

GuHuangAI, 2025-02-12:
> If you record the absolute values of the 6D pose, you need to calculate the delta of the 6D pose. Here, the 6D pose consists of XYZ (3) + Roll-Pitch-Yaw (3). We define delta translation as the difference in translation: Δ t = t n − t n − 1 . The definition of delta rotation involves first calculating the rotation matrices R n and R n − 1 , then computing Δ R = R n R n − 1 − 1 . Finally, convert the rotation matrix Δ R back to Euler angles.

Hello, I notice that robosuite uses the delta of axis-angle instead of euler-angle, and it applies a scale on the delta. 
As a comparison, CogAct uses the euler-angle without scaling.
I wonder know which setting is more useful, could you give some suggestion?