microsoft/CogACT #16: CogACT内存优化的问题

HUAFOR, 2025-01-28:
感谢您为具身社区带来出色的工作！
最近发现一个问题：在微调cogact的过程中发现机器内存一直在上升，在某个step会直接爆内存退出，如果机器内存越大爆内存的step会延后
想问下codebase是不是关于内存方面存在一些优化的空间，还是说是我们使用的问题？训练脚本大致如下：

torchrun --standalone --nnodes 1 --nproc-per-node 8 scripts/train.py \
  --pretrained_checkpoint "CogACT/CogACT-Base" \
  --vla.type prism-dinosiglip-224px+oxe+diffusion \
  --vla.data_mix $dataset \
  --vla.expected_world_size 8 \
  --vla.global_batch_size 256 \
  --vla.per_device_batch_size 32 \
  --vla.learning_rate 2e-5 \
  --vla.max_steps  $max_setp \
  --data_root_dir "***" \
  --run_root_dir "****" \
  --image_aug False \
  --save_interval 6000 \
  --repeated_diffusion_steps 8 \
  --future_action_window_size 15 \
  --action_model_type DiT-B \
  --is_resume False \

jasper0314-huang, 2025-01-28:
@HUAFOR I have encountered the same issue, with memory usage peaking at around 500G when using 8 GPUs.

arnoldland, 2025-02-09:
Thank you for your interest. This phenomenon is caused by TFDS. During training, the memory usage gradually increases until it eventually levels off at a constant value. If your shuffle_buffer_size is set too high and the system’s RAM is lower than this constant, the program will crash. (1 TB of RAM available on a single node with 8 A100 GPUs is sufficient.)

If you encounter this issue, you can reduce the``shuffle_buffer_size``in [CogACT/conf/vla.py](https://github.com/microsoft/CogACT/blob/main/conf/vla.py), although doing so might affect the model’s performance. Alternatively, you can create a PyTorch dataset and convert the dataset saved by TFDS into the appropriate format.

There are more detailed causes and discussions on this issue for reference:
[octo-models/octo#16](https://github.com/octo-models/octo/issues/16)
[openvla/openvla#4](https://github.com/openvla/openvla/issues/4)

closed 2025-02-20 by arnoldland