microsoft/CogACT #17: About Training Speed

jasper0314-huang, 2025-01-28:
Hi,

Thank you for sharing this excellent repo!

I’m trying to fine-tune CogACT on the BridgeV2 dataset using the script below. With 8 A100 GPUs, it takes about 6 minutes for 100 iterations. Is this an expected speed?

[image]

```bash
#! /usr/bin/env bash

export TF_CPP_MIN_LOG_LEVEL=3
export HF_TOKEN=XXX
export WANDB_API_KEY=XXX

torchrun --standalone --nnodes 1 --nproc-per-node 8 scripts/train.py \
  --pretrained_checkpoint CogACT/CogACT-Base \
  --vla.type prism-dinosiglip-224px+oxe+diffusion \
  --vla.data_mix "bridge" \
  --vla.expected_world_size 8 \
  --vla.global_batch_size 256 \
  --vla.per_device_batch_size 32 \
  --vla.learning_rate 2e-5 \
  --data_root_dir "./data/" \
  --run_root_dir "./experiments/" \
  --run_id 1234 \
  --image_aug True \
  --wandb_project CogACT \
  --save_interval 1000 \
  --repeated_diffusion_steps 8 \
  --future_action_window_size 15 \
  --action_model_type DiT-B \
  --is_resume False \
  --hf_token HF_TOKEN
```

Thanks in advance for your help!

arnoldland, 2025-02-06:
Thank you for your attention! A speed of approximately 3.32 seconds per iteration on 8 A100 GPUs with 80GB each is within the expected range.

jasper0314-huang, 2025-02-09:
> Thank you for your attention! A speed of approximately 3.32 seconds per iteration on 8 A100 GPUs with 80GB each is within the expected range.

@arnoldland Thank you for your response! I also wanted to ask about the GPUs used for evaluation and the approximate time required. 

For example, how long does it take to complete evaluation when running cogact_bridge.sh?

arnoldland, 2025-02-11:
Our experiments are mostly conducted on a single A6000 GPU, and testing confirmed that the A100 GPU can also run successfully. On a single A6000, running cogact_bridge.sh alone takes nearly one hour. We repeat the bridge experiment five times to calculate the average success rate.
In our actual experiments, we create multiple simulation environments simultaneously and synchronously concatenate the outputs of different simulators into a single batch. This batch is then used for a single inference with one policy model. This approach allows for more than a 50% increase in speed. The code for batch inference with the model can be found in [CogACT/vla/cogactvla.py](https://github.com/microsoft/CogACT/blob/main/vla/cogactvla.py#L390).

jasper0314-huang, 2025-02-16:
@arnoldland Got it. Thanks so much for your support!

closed 2025-02-16 by jasper0314-huang

CostaliyA, 2025-06-09:
> Our experiments are mostly conducted on a single A6000 GPU, and testing confirmed that the A100 GPU can also run successfully. On a single A6000, running cogact_bridge.sh alone takes nearly one [...]

I would like to ask how A100 runs the evaluation because it does not support ray tracing, and my A100 cannot run SimplerENV normally. Thank you for your reply.