microsoft/CogACT #19: run error killed

zmf2022, 2025-02-14:


whichwhichgone, 2025-03-14:
> Could you please provide more specific information? If the error occurred suddenly during training, it might be due to insufficient cpu memory, which is related to issue [#16](https://github.com/microsoft/CogACT/issues/16).

I found even I use a much smaller buffer size as suggested in #16 , the memory problem still exists. The code is run on a node of 8 A100 GPUs with 1T RAM.

arnoldland, 2025-03-15:
Could you kindly inform us of the exact ``shuffle_buffer_size`` and the approximate number of training steps at which the issue occurred? Additionally, if possible, could you provide details regarding the changes in memory usage over time or any other specific information?

whichwhichgone, 2025-04-16:
> Even when using `shuffle_buffer_size = 256,000` on the mixed oxe_plus_minus dataset, it will not exceed 1 TB of RAM in our exp. The same should be true for a single dataset. You can check to ensure that `num_worker = 0` is correctly set, so that parallel operations are properly handled by TFDS. Although with smaller data size, a smaller `shuffle_buffer_size` is feasible, we still expect that `shuffle_buffer_size = 128,000` can also run successfully on your setting. If there are still some problems, you can try reducing it to 100,000 for the single berkely_autolab_ur5 dataset.

Is there any difference with the package dependencies? I keep running into the out-of-memory error, even when I set the shuffle_buffer_size to 10,000. The memory just creeps up slowly like this until the whole program crashes

[image]

LukeLIN-web, 2025-04-27:
> > Even when using `shuffle_buffer_size = 256,000` on the mixed oxe_plus_minus dataset, it will not exceed 1 TB of RAM in our exp. The same should be true for a single dataset. You can check to ensure that `num_worker = 0` is correctly set, so that parallel operations are properly handled by TFDS. Although with smaller data size, a smaller `shuffle_buffer_size` is feasible, we still expect that `shuffle_buffer_size = 128,000` can also run successfully on your setting. If there are still some problems, you can try reducing it to 100,000 for the single berkely_autolab_ur5 dataset.
> 
> Is there any difference with the package dependencies? I keep running into the out-of-memory error, even when I set the shuffle_buffer_size to 10,000. The memory just creeps up slowly like this until the whole program crashes
> 
> <img alt="Image" width="1604" src="https://private-user-images.githubusercontent.com/166597188/434184900-a84f6f40-75f2-49d1-9b26-f3ca3fffd725.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU3MTMyNDEsIm5iZiI6MTc0NTcxMjk0MSwicGF0aCI6Ii8xNjY1OTcxODgvNDM0MTg0OTAwLWE4NGY2ZjQwLTc1ZjItNDlkMS05YjI2LWYzY2EzZmZmZDcyNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQyN1QwMDE1NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDIyYzQ5Nzc3NjAyODE4OTg2MDYyNzNmNzZlOWMwMzU2MWM2MDJlNTIwZjBiNTJjMjdhOTRlZTMyNzNkNThhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.qw0RC3eio_sjMlYPbsmH7Rfk6IQp7eKFE5kSwST8tG4">

because https://github.com/openvla/openvla/issues/4

arnoldland, 2025-04-28:
This seems quite strange. I checked the software package versions and didn't find any issues. Could you provide your software package dependency information? With such a small shuffle_buffer_size setting, this kind of memory behavior shouldn't occur. Has anyone else encountered a similar problem with a small buffer_size?

whichwhichgone, 2025-04-30:
> This seems quite strange. I checked the software package versions and didn't find any issues. Could you provide your software package dependency information? With such a small shuffle_buffer_size [...]

absl-py==2.1.0
accelerate==1.4.0
annotated-types==0.7.0
apache-beam==2.54.0
array_record==0.6.0
asttokens==3.0.0
astunparse==1.6.3
attrs==25.1.0
beautifulsoup4==4.13.3
blinker==1.9.0
cachetools==5.5.2
certifi==2025.1.31
charset-normalizer==3.4.1
chex==0.1.89
click==8.1.8
cloudpickle==2.2.1
coacd==1.0.4
contourpy==1.3.1
crcmod==1.7
cycler==0.12.1
debugpy==1.8.13
decorator==5.2.1
dill==0.3.1.1
distrax==0.1.5
dlimp @ git+https://github.com/moojink/dlimp_openvla@040105d256bd28866cc6620621a3d5f7b6b91b46
dm-reverb==0.14.0
dm-tree==0.1.9
dnspython==2.7.0
docker-pycreds==0.4.0
docopt==0.6.2
docstring_parser==0.16
draccus==0.10.0
einops==0.8.1
etils==1.12.0
exceptiongroup==1.2.2
executing==2.2.0
Farama-Notifications==0.0.4
fastavro==1.10.0
fasteners==0.19
filelock==3.17.0
flash_attn==2.5.5
Flask==3.1.0
flatbuffers==25.2.10
flax==0.8.1
fonttools==4.56.0
fsspec==2025.2.0
gast==0.6.0
gdown==5.2.0
gin-config==0.5.0
gitdb==4.0.12
GitPython==3.1.44
google-auth==2.38.0
google-auth-oauthlib==1.2.1
google-pasta==0.2.0
grpcio==1.70.0
gym==0.23.0
gym-notices==0.0.8
gymnasium==0.29.1
h5py==3.13.0
hdfs==2.7.3
httplib2==0.22.0
huggingface-hub==0.29.1
idna==3.10
imageio==2.37.0
imageio-ffmpeg==0.6.0
immutabledict==4.2.1
importlib_resources==6.5.2
ipython==8.22.2
itsdangerous==2.2.0
jax==0.4.34
jaxlib==0.4.34
jedi==0.19.2
Jinja2==3.1.5
Js2Py==0.74
json-numpy==2.1.0
jsonlines==4.0.0
jsonpickle==3.4.2
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
keras==2.15.0
kiwisolver==1.4.8
libclang==18.1.1
-e git+https://github.com/simpler-env/ManiSkill2_real2sim@ef7a4d4fdf4b69f2c2154db5b15b9ac8dfe10682#egg=mani_skill2_real2sim&subdirectory=../../ManiSkill2_real2sim
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline==0.1.7
mdurl==0.1.2
mediapy==1.2.0
mergedeep==1.3.4
ml-dtypes==0.2.0
ml_collections==1.0.0
mpmath==1.3.0
msgpack==1.1.0
mypy-extensions==1.0.0
namex==0.0.8
narwhals==1.28.0
nest-asyncio==1.6.0
networkx==3.4.2
ninja==1.11.1.3
numpy==1.24.4
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvcc-cu12==12.2.140
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-cusparselt-cu12==0.6.2
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
oauthlib==3.2.2
objsize==0.7.1
opencv-python==4.11.0.86
OpenEXR==3.3.2
openvla @ git+https://github.com/arnoldland/openvla@5603207085d55148682e2a35b868ad77d7b42ece
opt_einsum==3.4.0
optax==0.2.4
optree==0.14.0
orbax-checkpoint==0.4.4
orjson==3.10.15
packaging==24.2
parso==0.8.4
peft==0.11.1
pexpect==4.9.0
pillow==11.1.0
platformdirs==4.3.6
plotly==6.0.0
portpicker==1.6.0
promise==2.3
prompt_toolkit==3.0.50
proto-plus==1.26.0
protobuf==3.20.3
psutil==7.0.0
ptyprocess==0.7.0
pure_eval==0.2.3
pyarrow==14.0.2
pyarrow-hotfix==0.6
pyasn1==0.6.1
pyasn1_modules==0.4.1
pydantic==2.10.6
pydantic_core==2.27.2
pydot==1.4.2
pygame==2.1.3
Pygments==2.19.1
pyjsparser==2.7.1
pymongo==4.11.1
pyparsing==3.2.1
PySocks==1.7.1
python-dateutil==2.9.0.post0
pytz==2025.1
PyYAML==6.0.2
pyyaml-include==1.4.1
referencing==0.36.2
regex==2024.11.6
requests==2.32.3
requests-oauthlib==2.0.0
rich==13.9.4
rlds==0.1.8
rpds-py==0.23.1
rsa==4.9
Rtree==1.3.0
ruckig==0.14.0
safetensors==0.5.3
sapien==2.2.2
scipy==1.12.0
sentencepiece==0.1.99
sentry-sdk==2.22.0
setproctitle==1.3.5
simple-parsing==0.1.7
-e git+https://github.com/simpler-env/SimplerEnv@052eaea54dc7c37447cf40c499134b55534bbde4#egg=simpler_env
six==1.17.0
smmap==5.0.2
soupsieve==2.6
stack-data==0.6.3
sympy==1.13.1
tabulate==0.9.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-addons==0.23.0
tensorflow-datasets==4.9.3
tensorflow-estimator==2.15.0
tensorflow-graphics==2021.12.3
tensorflow-hub==0.16.0
tensorflow-io-gcs-filesystem==0.37.1
tensorflow-metadata==1.16.1
tensorflow-probability==0.23.0
tensorstore==0.1.45
termcolor==2.5.0
tf_keras==2.15.1
tfds-nightly==4.9.4.dev202403220044
tfp-nightly==0.25.0.dev20240322
timm==0.9.10
tokenizers==0.19.1
toml==0.10.2
toolz==1.0.0
torch==2.6.0
torchaudio==2.6.0
torchvision==0.21.0
tqdm==4.67.1
traitlets==5.14.3
transformers==4.40.1
transforms3d==0.4.2
trimesh==4.6.3
triton==3.2.0
typeguard==2.13.3
typing-inspect==0.9.0
typing_extensions==4.12.2
tzlocal==5.3
urllib3==2.3.0
wandb==0.18.7
wcwidth==0.2.13
Werkzeug==3.1.3
wrapt==1.14.1
zipp==3.21.0
zstandard==0.23.0

arnoldland, 2025-05-06:
I’m using draccus==0.8.0, triton==2.2.0, and torch==2.3.1. However, I don’t think these packages are causing the bug you mentioned—the root of the problem is likely related to TFDS. You might want to check out #16 and consider opening an issue in the repository mentioned there to seek further help.