microsoft/CogACT #20: Warning about `padding_side`

GuHuangAI, 2025-02-16:
When I used the Cogact-base model for inference, there was an warning:
`A decoder-only architecture is being used, but right-padding was detected! For correct generation results, please set `padding_side='left'` when initializing the tokenizer.`
What should I do to remove this warning?

arnoldland, 2025-02-18:
Thanks for your question! If you encounter this warning when using the ``predict_action_batch`` function, it will not affect the model's correct inference, even when processing in batches. We do use right padding and will only perform a single forward pass through the LLM to obtain the necessary tokens. Regardless of whether left or right padding is used, as long as it is implemented correctly, the results will be the same. The reason this warning appears is that, for simplicity of the code, LLMs typically prefer to use left padding.

GuHuangAI, 2025-02-18:
Thanks for your reply.

closed 2025-02-20 by arnoldland