microsoft/CogACT #25: Evaluation Expansion Application

MstarLioning, 2025-03-20:
Hello, thank you for your work that has given me new inspiration! May I ask if you have any plans to evaluate your model on LIBERO? I look forward to your reply!

arnoldland, 2025-03-22:
Thank you for your interest and questions! We believe that the Simpler environment is closer to real-world testing compared to LIBERO. Moreover, LIBERO's evaluation does not emphasize the model's generalization capabilities, which is why we primarily use Simpler for evaluation. Currently, I don't have plans to conduct evaluations on LIBERO, but if you're interested, we warmly welcome you to have a test and contribute! Additionally, in other works, our method has been compared across other simulators, which you might find useful as a reference.

MstarLioning, 2025-03-23:
Thank you for your reply, I understand what you mean! Due to the fine-tuning required in the previous VLA evaluation, I think I will try evaluating CogACT on LIBERO once I have sufficient computational resources.

closed 2025-03-23 by MstarLioning