microsoft/CogACT #27: About collecting custom fine tuning data

JaehongJaehongMin, 2025-04-14:
Hello!
First of all, thank you for sharing your great work!
I was impressed that CogACT performs well on SimplerEnv and LIBERO simulation, even without additional fine-tuning, unlike some other models.

I’m planning to deploy CogACT on a UR10e robot arm in our lab, as it showed the best performance among the models I've reproduced. I read your advice in the README that collecting our own fine-tuning dataset is recommended for deployment in a new environment.

I have one question regarding that:
Do I need to align the UR10e's coordinate axes with a specific convention? Or is it enough to just ensure that the relative alignment between the robot and the camera is consistent between fine-tuning and deployment?

Thank you in advance!

arnoldland, 2025-04-15:
Many thanks for your recognition and the efforts invested in reproducing the results! 
The discussion here involves two kinds of demands:

1.​​ ensuring the satisfactory task performance​​.
For a new robot and environment, we observed that maintaining a fixed relative position and orientation between the camera and the robot while aligning actions into the camera coordinate system during both training and prediction phases already yields satisfactory performance. The camera placement can be roughly configured in a first-person perspective similar to Google Robot (numerical precision is unnecessary), as demonstrated in ​​our Appendix Figure I​​. Notably, even with deviations in camera angles (e.g., the configuration in ​​our Appendix Figure II​​), the system remains robust.
2. exploration of the optimal camera and robot configuration​​.
Achieving optimal performance requires considering additional factors: leveraging pre-trained data effectively, minimizing occlusions in camera-robot positioning, etc. If the target robot already has sufficient instances in the pre-training dataset, aligning configurations with those in the training data is preferable.

I understand your primary focus is on ​​Point 1​​, which means you just need ensuring consistent relative positioning between the robot and camera during both fine-tuning and evaluation phases. This alignment is critical for maintaining action-space consistency across training and deployment environments

JaehongJaehongMin, 2025-04-17:
Thank you very much! Your advice has been really helpful.
May I ask one more question?

I saw your comment in another issue mentioning that CogACT uses Euler angles.
However, I believe our UR robot operates based on axis-angle representation.
In that case, should I convert the Euler angle output from CogACT into axis-angle before sending commands to the robot arm?

I assume CogACT learns better from real-world fine-tuning data using Euler angles, so I'm planning to collect the data in Euler format. Does this sound like the right approach?

arnoldland, 2025-04-19:
1. Q: Is it aligned with world frame or tcp frame?
Our Euler angles are represented in a fixed coordinate system, which can be considered closer to a world frame. In practice, both the fine-tuning data and testing for our real-world robots involve converting all actions to a fixed camera coordinate system.

2. Q: Does this sound like the right approach?
Yes, you are right. We recommend first converting the axis-angle representation training data to Euler angles for fine-tuning CogACT, then converting the Euler angle rotations output by CogACT back to axis-angle representation. This approach is intended to ensure unified representation of the model's output across different robots, while actual deployment only requires conversion to the specific control mode.

arnoldland, 2025-04-28:
The FMB dataset we downloaded is indeed 356GB. Perhaps you can check if there has been a version update of the FMB dataset that caused the increase in size. However, I checked the original FMB dataset repository, and the reported total size is still less than 1TB.

JaehongJaehongMin, 2025-05-01:
Thank you very much for your kind explanation.
I'm sorry for asking so many questions. I'm still quite new to the VLA field.

Before collecting real-robot trajectories, I'm trying to gather trajectories in the SimplerEnv simulation environment, which is mentioned in your paper.
I collected **25 trajectories for each of the 4 tasks** (Carrot, Cube, Spoon, Eggplant) in SimplerEnv, trying to ensure as much diversity in episodes as possible.
The trajectories contain images (480*640 rgb) and actions (7-DoF (delta x, delta y, delta z, rot x, rot y, rot z, gripper), rotations are in Euler format).
I then converted the collected trajectories to the RLDS format, following this repository: (https://github.com/kpertsch/rlds_dataset_builder).
Finally, I fine-tuned CogACT-BASE with the collected trajectories (converted to RLDS format). I'll refer to this as **my data**.
However, the success rates were **all zero** (sadly...).

Meanwhile, I obtained another fine-tuning dataset (I'll call this **other_repo data**) from a different repository that was used to fine-tune OpenVLA..
It worked well, so I fine-tuned CogACT with this other_repo data, and the performance improved.

I analized the other_repo dataset, and I found that they qunatized the action space (possibly for tokenization?).
So I applied quantization to my data as well and fine-tuned CogACT. The success rates were not zero, but still lower than default CogACT-BASE.
I think I may have made mistake when preprocessing the data (quantizing the action space of **my data**).
Could you please provide me with any advice on constructing a fine-tuning dataset?

JaehongJaehongMin, 2025-05-01:
I'm attaching my training loss curve.
The pink one is **other_repo data**, and the purple one is **my data**.

I stopped training at 3000 steps in case of **my data**, because the loss didn't change after that.

[image]

JaehongJaehongMin, 2025-05-05:
I'm attaching the histogram of my data (not quantized) and other_repo data.
The first row is my data and the second row is other_repo data.

[image]

arnoldland, 2025-05-06:
It's OK to discuss this here!
First of all, I don't think quantized data is necessary for CogACT, because diffusion models are able to model continuous distributions (but you need to normalize data).
Secondly, has the model been fine-tuned for a sufficient number of steps? The loss in diffusion models typically drops rapidly at first, followed by a long period of slow decline. So even if the diffusion loss appears to decrease slowly over a short period, it doesn't necessarily mean the training is sufficient (for example, training for just a few thousand steps may not be enough).
Additionally, if there's concern about the quality of the collected data, you can replay the collected ground truth data in the simulation environment to verify whether the task can still be completed.

JaehongJaehongMin, 2025-05-06:
I'm really thankful for your reply. It helps me a lot.

> First of all, I don't think quantized data is necessary for CogACT, because diffusion models are able to model continuous distributions (but you need to normalize data).
I admit with you! CogACT can predict action in continuous space, which means doesn't need to quantize it.
Actually, I didn't normalized the actions, because the **other_repo** data shows good performance even if they are not normalized.
Do you have any idea with it that **other_repo** data shows good performance without normalization?
By the way, I think I need to try training with my data after normalization.
Is it enough to normalize the actions **for each dimension from q01 to q99** with **clipping the outlier** (outside the q01 and q99)?
(And I guess I should save the statistics before normalization for inference in simulation, because they use very small number for actions. It seems that they ignore the big number, for example bigger than 0.5)

> Secondly, has the model been fine-tuned for a sufficient number of steps?
I've trained it for 5000 steps with 128 global batches (16 batches per gpu for 8 gpus) about 200 epochs with 2e-5 lr.
I'm attaching my loss curve below. The one is **my data** (not quantized, poor performance), and the one is **other_repo** data (quantized, good performance).
**other_repo** data show good performance even with 5000 steps (about 200 epochs).
But I should try more steps, because it could be easy to learn quantized actions with smaller steps.

[image]

> Additionally, if there's concern about the quality of the collected data, you can replay the collected ground truth data in the simulation environment to verify whether the task can still be completed.
I've already checked that my data and other_repo data works well in replay, but I think it's a really important point! Thank you very much.

Thank you for all of your advice. I'll try training CogACT with **quantized** (but save the original statistics like q01 and q99 for inference in Simpler) and **more steps**!

arnoldland, 2025-05-06:
Finetuning the entire network with only 5K steps is indeed too few — a longer finetuning process (at least 10K steps) is needed, and you’ll observe that the loss is still decreasing.

CogACT-Base itself is capable of handling tasks on simpler. Therefore, if you finetune directly with few steps and without normalization, it might interfere with actions that would be correct in the normalized space, potentially leading to performance degradation. Moreover, such a distribution of values may be unfavorable for the diffusion to learn effectively.

During training, you need to save the statistics and restore the predictions from the normalized values when evaluating in the simulation environment. I’m not quite sure why the model seems to perform better after quantization, but according to what you previously mentioned, quantizing your collected data did not lead to better performance. Perhaps the performance gain in the other repo comes from different factors? For example, is there any special technique in how they collected their data?

JaehongJaehongMin, 2025-05-06:
Sorry the **quantized** at the last line in my previous feed was just a typo. It was **normlized** (I fixed it).
I'll try training with normalization.

> a longer finetuning process (at least 10K steps) is needed

I see... I should try it with more steps!

> During training, you need to save the statistics and restore the predictions from the normalized values when evaluating in the simulation environment.

That is the thing that I really concerned. Thank you!

> is there any special technique in how they collected their data?

I'm trying to find out anything related with it, but I couldn't. I'm struggling with it almost about 2 months.

Sorry, the **quantized** at the end of my previous message was a typo — I meant **normalized** (I’ve fixed it).
I’ll try training with normalization.

> a longer finetuning process (at least 10K steps) is needed

I see — I should try with more steps!

> During training, you need to save the statistics and restore the predictions from the normalized values when evaluating in the simulation environment.

That was exactly what I was concerned about. Thank you!

> Is there any special technique in how they collected their data?

I’ve been trying to find any related information, but haven’t had much luck. I’ve been struggling with this for almost two months. 😢

JaehongJaehongMin, 2025-05-06:
Sorry, the quantized at the end of my previous message was a typo — I meant normalized (I’ve fixed it).
I’ll try training with normalization.

    a longer finetuning process (at least 10K steps) is needed

I see — I should try with more steps!

    During training, you need to save the statistics and restore the predictions from the normalized values when evaluating in the simulation environment.

That was exactly what I was concerned about. Thank you!

    Is there any special technique in how they collected their data?

I’ve been trying to find any related information, but haven’t had much luck. I’ve been struggling with this for almost two months. 😢