microsoft/CogACT #30: Does ensemble aggregate two action or three actions?

LukeLIN-web, 2025-04-27:
https://github.com/microsoft/CogACT/blob/d6c529f465653364a26ff1ec7e81c8d8c7976531/sim_cogact/adaptive_ensemble.py#L19

In my understanding ,         `self.pred_action_horizon = pred_action_horizon        self.action_history = deque(maxlen=self.pred_action_horizon)  `  

seems that         self.action_history.append(cur_action)
        num_actions = len(self.action_history) max = 2, 
only aggregate two actions? 

But in paper the figure looks like it aggregate three actions. Figure 3. Illustration of our action ensemble strategy with K = 2
(using the last 2 historical action predictions

arnoldland, 2025-04-28:
Hello, we indeed aggregate two actions on the Google robot. The Figure 3 in the paper is just an example, showing a schematic diagram for the case when K=2, and not the actual parameters used. We apologize for the misunderstanding here.

closed 2025-05-06 by arnoldland

LukeLIN-web, 2025-05-12:
> Hello, we indeed aggregate two actions on the Google robot. The Figure 3 in the paper is just an example, showing a schematic diagram for the case when K=2, and not the actual parameters used. We apologize for the misunderstanding here.

I have a problem .  In the paper is said , `(N = 15 by default), leading to a context length of  N + 2 = 17`

Why it +2? K=2  seems only add one more action?

arnoldland, 2025-05-19:
The number of predictions during training does not need to match the number used for decision-making during execution. For cross-embodiment tasks, using 16 steps during training aims to accommodate the robots with the highest control frequency. During actual inference, the number of predicted actions should be adjusted according to the control frequency or range of the deployed robot. For example, the Google robot operates at a control frequency of 3 Hz—using 16 steps would mean predicting all actions for the next 5 seconds at once, which would result in a very low feedback frequency.

arnoldland, 2025-05-21:
Apologies for the confusion in my previous response. I understand your question now. 
N=15 refers to predicting actions for the next 15 steps, while the action of the current step also needs to be predicted, hence we add 1. Additionally, there's a cognition token that must be fed into the action module, leading the context length of the action module to be increased by another 1. This results in the 15+2 calculation. 
Regarding the K you mentioned -- this is separate from the number of actions predicted during training; it's the number of actions the model references when generating the actual executed action.