microsoft/CogACT #38: Regarding Fine-tuning of the model for specific task

sidharth72, 2025-06-24:
Hi there, the work is wonderful,

I'm trying to run CogACT in another environment, such as Gazebo, but it is not performing as expected as it does in SIMPLER. So, what I thought is that the pre-trained model is provided in the Huggingface Repo, and do we need to fine-tune it again to achieve an out-of-the-box experience? Or does the model have a general understanding to perform a task with any robot? I'm currently trying with the Widow X robot in Gazebo for picking coke can and some other tasks as well, I'm skeptical about whether the model can do the task like this without fine-tuning, because in the SIMPLER env for evaluation, I've only seeing Google Robot doing the same, not the widowx.

Thank you in advance!

arnoldland, 2025-06-30:
Thank you for your recognition! If you're running in Gazebo, it is necessary to fine-tune the model to accomplish the task. Our model does not have the capability to generalize to a robot it has never seen before. As far as I know, there is no VLA that can achieve comparable task success rates on completely unseen robots. Different robots may have different degrees of freedom, control schemes, and other variations, all of which make it difficult to deploy the model in a zero-shot manner on a new robot.

sidharth72, 2025-07-08:
> Thank you for your recognition! If you're running in Gazebo, it is necessary to fine-tune the model to accomplish the task. Our model does not have the capability to generalize to a robot it has [...]

Thank you for shedding light on my confusion!

closed 2025-07-08 by sidharth72